eldruin/ads1x1x-rs #4: Setting the address

David-OConnor, 2020-06-21:
Hi. How do you set alternate addresses, using the convention described in the datasheets? Eg for the ADS1115, it has 4 address possibilities: Address pin connected to:
- GND (0x48, default)
- VCC
- SDA
- SCL

This doesn't seem to line up with the API in a way I can find in the docs:
```rust
let (a1, a0) = (true, false);
let address = SlaveAddr::Alternative(a1, a0);
let adc = Ads1x1x::new_ads1013(dev, address);
```

eldruin, 2020-06-21:
The 4 addresses in the [datasheet](https://www.ti.com/lit/ds/symlink/ads1115.pdf#%5B%7B%22num%22%3A716%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C602.2%2C0%5D) are already representable as:

| Address Pin Connection | Effective Address | Representation |
|-----|-|-|
| GND | `1001000` | `SlaveAddr::default()` or `SlaveAddr::Alternative(false, false)` |
| VCC | `1001001` | `SlaveAddr::Alternative(false, true)` |
| SDA | `1001010` | `SlaveAddr::Alternative(true, false)` |
| SCL | `1001011` | `SlaveAddr::Alternative(true, true)` |

Am I missing anything?

David-OConnor, 2020-06-21:
Thank you! That answers my question. Submitted a PR to switch to a more explicit system.

closed 2020-06-21 by David-OConnor