eldruin/ads1x1x-rs #7: ADS1115 read return always 32767

simeondmr, 2021-01-16:
Hello,

I'm using ADS1115 and when I call read method like this: block!(self.adc.read(&mut channel::SingleA0));
the function return always: 32767, and this is really strange.

When I call the read method like this: block!(self.adc.read(&mut channel::DifferentialA0A1));
it seems to work correctly.

The dependencies that I'm using are:

ads1x1x = "0.2.1"
i2cdev = "0.4.4"
embedded-hal = "0.2.4"
linux-embedded-hal = "0.3.0"
nb = "1.0.0"

Please could you help me with this problem?
Thanks

Simeon

eldruin, 2021-01-18:
What you describe is possible depending on how your pins are electrically connected and your choice of full scale range.
Please try building a voltage divider as described here:
https://blog.eldruin.com/ads1x1x-analog-to-digital-converter-driver-in-rust/
Remembering to add the line `adc.set_full_scale_range(FullScaleRange::Within6_144V).unwrap();` for 5V.
And then see if you have the same problem.

simeondmr, 2021-01-18:
Hello,

Ok, I will build a voltage divider and I will let you know!
Thanks .

Simeon

simeondmr, 2021-01-18:
Hello,

I have done the voltage divider and the code work correctly.
The problem was that I forgot the line: adc.set_full_scale_range(FullScaleRange::Within6_144V).unwrap();
Thanks for your time.

Simeon

closed 2021-01-18 by simeondmr