eldruin/ads1x1x-rs #9: toggling FullScaleRange?

pdgilbert, 2021-04-30:
(Maybe really a hardware question.)

I am using ads1x1x-rs to read differential voltage across low-side shunt resistors to give a rough estimate of load current and battery charge/drain. There is also a solar panel which is not being measured.  I've tested with a bluepill and blackpill stm32f411 and it works reasonably. (Maybe even better than should be expected given that I haven't bothered with op amps or capacitors to calm the noise.) For this I am setting `FullScaleRange::Within0_256V`.

I am wondering about also measuring the high side voltage, for which I would want `FullScaleRange::Within4_096V`.  Is it reasonable to consider toggling `FullScaleRange` between these settings when I move to different pins in my read loop, or is that asking for real trouble? I've read the datasheet but didn't find anything useful on this. (They sometimes skip stupid novice details.)

eldruin, 2021-05-01:
Changing the full scale range should be fine. I would say what you should be careful about is if you are mixing power sources like a battery and your solar panels. However, I do not know enough about electronics to give detailed advice.

pdgilbert, 2021-05-03:
Thanks, yes this does work with no ill effects that I have seen so far. (The solar panel feeds through a battery charging module first, so it is near battery potential, especially when actually connected to the battery. That does not seem to be causing problems either.)

closed 2021-05-03 by pdgilbert