eldruin/ads1x1x-rs #12: ADS1119 support?

clintfred, 2023-09-01:
Thanks for the great crate! (and all the other ones as well)

Do you know if this library supports or could support the [ADS1119](https://www.ti.com/product/ADS1119)? The numbering scheme makes it seem like it might be compatible, but that may be misleading.

We were trying to use 2 ADS1115, but we ran into I2C address conflict issues and the EE I am working with suggested we use the 1119 instead.

I'm fairly new to the embedded world, so any guidance is appreciated.

eldruin, 2023-09-04:
You are welcome. I'm glad they are useful to you.
The ADS1119 is not immediately compatible with this driver as it features a command-based interface and the register map also differs.
The device seems fairly simple, though, so you may be able to write a driver for it.

Nevertheless, if you are only using 2 ADS1115, you should be able to just change the polarization of the `ADDR` pin to change its address and avoid the conflict. Actually, you can have up to 4 ADS1115 connected to the same bus simultaneously. See [Table 4 in section 9.5.1.1 of the datasheet](https://www.ti.com/lit/ds/symlink/ads1115.pdf#%5B%7B%22num%22%3A716%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C602.2%2C0%5D).

clintfred, 2023-09-11:
Thanks for the reply. 
Would it make sense to try to add the 1119 to this crate, or is it different enough that that wouldn't be appropriate? I may be interested in attempting to add 1119 support if I can't find any alternatives.

(After talking to the EEs, I've learn the ADS1115 were also conflicting with other I2C devices as well. So we're forging ahead with using the 1119.)

eldruin, 2023-09-12:
The ADS1119 is actually quite different so I think a separate driver would make more sense.
Please note that you can also use a PCA9548A or similar device to switch between devices in your I2C bus and resolve your conflicts.
I have a [driver for a few of them here](https://crates.io/crates/xca9548a).

clintfred, 2023-12-18:
I have created a partial driver. You can find it [here](https://github.com/cambrianworks/ads1119-rs)

Any comments would be welcome. :)

eldruin, 2023-12-20:
Great. I do not have time to review it but let me know if you need help with any specific issue.
Closing.

closed 2023-12-20 by eldruin