eldruin/ads1x1x-rs #25: Release to crates.io to allow embedded-hal 1.0 compat

democat3457, 2024-04-14:
Would it be possible to make a release to be able to use this library as a crates.io dependency that doesn't break dependencies on embedded-hal v1.0? Right now, I am using a clone of this library locally and putting the path in my Cargo.toml, but that feels a little strange.

eldruin, 2024-04-29:
Thanks for the heads up. At the moment there are a couple of PRs that need thorough review and I would like to go over those before publishing a new version.

eldruin, 2025-01-14:
Sorry for the delay, everybody.
I have now published [version 0.3.0](https://github.com/eldruin/ads1x1x-rs/releases/tag/v0.3.0) which should solve this. :tada: 
Closing.

closed 2025-01-14 by eldruin