eldruin/ads1x1x-rs #26: The `use_alert_rdy_pin_as_ready` does not enable pulse in continuous-conversion mode as documented

Slabity, 2024-04-24:
I've been trying to get an ADS1115 to emit an interrupt signal when in continuous-conversion mode using this crate. According to the documentation, the `use_alert_rdy_pin_as_ready()` function says:

> This the ALERT/RDY pin outputs the OS bit when in OneShot mode, and provides a continuous-conversion ready pulse when in continuous-conversion mode.

However, this doesn't seem to work as expected. According to the TI documentation here: https://www.ti.com/lit/ds/symlink/ads1115.pdf

> Set the COMP_QUE[1:0] bits to any 2-bit value other than 11 to keep the ALERT/RDY pin enabled, and allow the conversion ready signal to appear at the ALERT/RDY pin output.

But the function itself calls `self.disable_comparator()` which sets the `COMP_QUEUE` value to `0b11`. This disables pulsing, and I need to manually call `set_comparator_queue(ComparatorQueue::One)` to set those bits low and and enable it again.

I'm not sure if this is an issue in the function itself or the documentation.

eldruin, 2024-05-02:
Interesting. I guess I might have understood it backwards at the time.
Could you send a PR changing the `disable_comparator` call there to `set_comparator_queue(ComparatorQueue::default())` in case the `COMP_QUEUE` is set to `0b11` and update the function documentation and [tests](https://github.com/eldruin/ads1x1x-rs/blob/0ff4c7b448fd388015d4903b774c63991356b954/tests/tier2.rs#L126-L161)?

Slabity, 2024-05-05:
> Interesting. I guess I might have understood it backwards at the time.

I just noticed that line seems to be a change that was added in a newer doc. Not sure if you might have been looking at an older revision.

I can make those changes in a few days.

eldruin, 2024-07-31:
Fixed in #27

closed 2024-07-31 by eldruin